makerbot/ReplicatorG #303: download link broken for mac binary/installer

n8cn, 2020-04-14:
http://replicat.org/download links to http://replicatorg.googlecode.com/files/ReplicatorG-0040-Installer.dmg which doesn't exist.

gallopinggoose6, 2020-04-14:
I would highly recommend finding an alternative to replicatorg, as the project appears to be completely abandoned by this point, as more powerful tools out there exist. Out of curiosity, what machine are you trying to use?

n8cn, 2020-04-15:
Thanks,
I came to that conclusion myself after I installed replicatorG from the Google Code repo (same version, link was not broken there) and it wouldn't even run without the Legacy Java 6 support installed which isn't available for MacOS Mojave/Catalina.

I've got an i3 clone and a mac running mojave and am new to 3d printing. Google results seemed to indicate that ReplicatorG is (or was) the best solution for this. It ended up not being usable for me. 

:/

gth, 2021-11-15:
ReplicatorG is still buried the Google Code Archive area -
https://code.google.com/archive/p/replicatorg/downloads

But as @gallopinggoose6 mentioned above, ReplicatorG is ancient, neglected software by today's standards  (I came here looking for some old Replicator 2X stuff :).